retro16/acsi2stm #87: Adding external activity LED on Mega STE internal version

fdanapfel, 2025-03-06:
Hi,

I'm using the internal version of acsi2stm for the Mega STE (https://github.com/retro16/acsi2stm/tree/stable/pcb/MegaSTE%20Internal ) which unfortunately doesn't seem to have a connector to connect the LED from the Mega STE disk cover to be able to see disk activity when the case is closed.

Is there a possibility to add a connector for the LED or can cables from the LED be soldered on the STM32 directly somehow?

tophe38, 2025-03-10:
Hi,

Bluepill HDD led activity come from PC13, which is Pin2 on the bluepill header.
But PC13 is the negative of the LED, other side should be connected to VCC3V3 (Pin 18)
Here is the LED schematic (from Bluepill schematic)

[image]

retro16, 2025-03-24:
Hi @tophe38, I never owned nor seen a real Mega STE, so feel free to disregard my message if it's silly.

Perhaps you could add a connector for the external LED: add another resistor and the correct connector for the external LED (I suppose it is standard "Dupont" pins ?) between VCC and PC13.

That would be a very cool addition to the official Mega STE PCB you produced.

JM

tophe38, 2025-03-25:
Hi,
It's done, but I did not push it, since it wasn't a big/important change.
But it makes sense, so I'll see tonight to make pull request.
The HDD led is a simple 2 wires with 2.54 female connector.
Thanks for the request.