retro16/acsi2stm #89: Fastload flag in Gemdrive mode not handled correctly

czietz, 2025-03-23:
Looking at the source code, in Gemdrive mode the BSS is _only_ cleared when the fastload flag in the PRG header is _not_ set:
https://github.com/retro16/acsi2stm/blob/aae0f6097b89e1470ef4b23484c8e42bbfd1f8e6/acsi2stm/GemDrive.cpp#L2269-L2271

Unfortunately, this is incorrect. The BSS must always be zeroed. The fastload flag decides whether the rest of the memory – after the BSS and up to p_hitpa – is cleared. See the description in the TOS 1.04 release notes:
[image]

... or the implementation in Hatari:
https://github.com/hatari/hatari/blob/d635e360dcc783ec971408d703c978103245f867/src/gemdos.c#L4319-L4325
https://github.com/hatari/hatari/blob/d635e360dcc783ec971408d703c978103245f867/src/gemdos.c#L4346-L4357

... or the implementation in TOS:
https://github.com/th-otto/tos3x/blob/b7b637339a1d3addf20524f046b528abe11d7cb9/bdos/proc.c#L675-L683

Not clearing the BSS (even though it must always be done, regardless of the fastload flag) causes a multitude of issues with programs.

retro16, 2025-03-24:
Hi Christian,

Thank you *so much* for this valuable feedback. I have limited knowledge about TOS, so it's always very welcome to have such hints to improve the product.

That looks like a quick fix, that will be part of the next release (which will be a stable release). I don't have much time currently, it may take weeks.

JM

retro16, 2025-03-24:
Implemented in 5.1b. Didn't fully test, but it seems that it's working.

czietz, 2025-03-24:
Confirmed to be fixed with commit d0b1298f818eb6d06f0754296209b6f107ea9576. Thank you!

closed 2025-03-24 by czietz