retro16/acsi2stm #90: Gemdrive mode freezes system when Alt-RAM is present

czietz, 2025-03-23:
On an ST or STE with Alternate RAM (provided by addons such as the MonSTEr, MonSTerbo, Magnum ST, Storm ST, Twister, ...) accessing the Gemdrive freezes the system to the point where even Ctrl-Alt-Del doesn't work anymore.

Note that you cannot use the DMA to access Alt-RAM, as the DMA only works in ST-RAM. Therefore, on a system with Alt-RAM there is a so-called "fast-RAM buffer" in ST-RAM, the address of which can be found via the _FRB cookie. This is used as temporary buffer for DMA by TOS, hard disk drivers, etc.

A quick search through the ACSI2STM code base found not reference to the _FRB cookie; therefore, I assume – without knowing with certainty – that this is the cause of this issue.

czietz, 2025-03-23:
Upon closer inspection, ACSI2STM even has a check for DMA-compatible memory:

https://github.com/retro16/acsi2stm/blob/aae0f6097b89e1470ef4b23484c8e42bbfd1f8e6/acsi2stm/SysHook.h#L378-L389

However, the address range is too broad. As mentioned, the DMA can only access ST-RAM. On a ST/MegaST/STE/MegaSTE, this means the first 4 MiB of memory. (And on a TT the ACSI-DMA is restricted to ST-RAM, too, which can be max. 10 MiB there.)

retro16, 2025-03-24:
Fantastic feedback, again ! I can't thank you enough for your insight.

Unfortunately, I don't have enough ROM space left on the STM32 to implement the horribly complex logic required to read a cookie correctly.

Here, it says "max 14MB", which is a widely known value: https://temlib.org/AtariForumWiki/index.php/Exception_vectors_and_basic_RAM_%26_ROM_locations
14MB would translate to phystop = 0xd00000, that would be a simple fix, but probably not enough for these expansions to work.

I have no extra hardware and only own basic ST / STE machines, so I have no way to test these expansions.

For now, a possible workaround would be to use the PIO firmware variant which doesn't do DMA at all. The code forces "isDma" to "true" in PIO mode, but DMA is simulated using CPU reads/writes in that mode.

JM

czietz, 2025-03-24:
Only the Falcon can have 14 MiB of (ST-)RAM. But the Falcon doesn't have an ACSI port, and is therefore not a target for the ACSI2STM. Maybe you could read the actual phystop system variable instead of hard-coding a value.

retro16, 2025-03-24:
Some tools alter the phystop value, but at this point it may be better to actually rely on it. After all, access works in all cases, it's just much slower if outside of the DMA range.

Thanks a lot for the info, I will try to implement phystop read if I have enough ROM space.

JM

retro16, 2025-03-24:
Implemented in 5.1b. Didn't fully test, but it seems that it's working.

czietz, 2025-03-24:
Thank you. However, as of commit d0b1298f818eb6d06f0754296209b6f107ea9576 this is _not_ fixed. Loading GEMDRIVE.PRG still hangs the system when Alt-RAM is present.

Flashing the PIO mode firmware and using GEMDRPIO.PRG works, though, even with Alt-RAM. So there appears to be some issue with the DMA, still. I'll try figure it out, but can't promise when I'll have time for debugging.

czietz, 2025-03-25:
At first glance it appears that it now hangs whenever a program runs in Alt-RAM and makes its first GEMDOS call. In that case, the stack and therefore parameters to the GEMDOS call are in Alt-RAM. I wonder: Do you read the GEMDOS opcode and parameters from stack using the DMA?

retro16, 2025-03-26:
Nice catch ! You found the root cause of the issue. That's exactly what I do:

https://github.com/retro16/acsi2stm/blob/d0b1298f818eb6d06f0754296209b6f107ea9576/asm/GEMDRIVE/syshook.s#L34-L61

On every system call, I set the DMA pointer address to the parameters, then I send the single-byte GemDrive command to the STM32 (it calls GemDrive::onGemdos), and the STM32 just reads parameters to decide whether it handles the call or forwards it to the system.

Now the hard part:
* This ASM code must fit inside a boot block (~500 bytes plus some header)
* This code is highly performance critical
* I don't have enough room to hold multiple variants of that code

The PIO variant degrades performance much more because it does a simulated DMA transfer, I didn't optimize PIO code at all.

There aren't tons of alternatives, here are my current ideas (all of them are heavy to implement and not so desirable):
* Pass the stack pointer at a fixed ST-RAM address and use the standard functions to read parameters (functions that honor dmatop). That will make the critical path much slower even on a standard ST.
* Make the PIO variant able to respond to the boot sector query (for convenience) and optimize PIO for speed. That would also work around MegaSTE cache issues (root cause unknown) and TT-RAM issues (same problem as Alt-RAM). PIO has the potential to be nearly as fast as DMA on faster machines if the code is correctly optimized.